davidepatti/noxim #148: Reading buffer values inside Router Rxprocess functions is not accurate.

cgraider, 2022-11-18:
Hi.
(i'm using accessnoxim)
i'm trying to cout buffer.size inside functions of rxprocess(which is called by [SC_METHOD]) and trying to see the increasing rate of the function. 
when i use cout buffer.size inside rxprocess i get accurate result which is increment/decrement by 1 but when i cout inside of functions defined in rxprocess (routingfunction for example) i get skipped results like 0 1 2 6.

why this happens?

davidepatti, 2022-12-07:
I don't understand what do you mean by "functions defined inside rxProcess()". 
rxProcess() is a method, that calls other methods (functions)
So when  you invoke "cout" from other functions, then they just maybe are NOT executed at each cycle...
By the way, can you give me a practical example of two lines in the following file where putting the "cout" gives this effect? 
[REFERE TO THIS FILE](https://github.com/davidepatti/noxim/blob/master/src/Router.cpp)

cgraider, 2023-03-28:
My mistake, i was putting cout inside Routing function which it doesn't get called every cycle it only get called when there is a input packet.

closed 2023-03-28 by cgraider